sandeepmistry/arduino-LoRa #579: Work with Helium

kargarisaac, 2022-05-24:
Hello,

Is it possible to use this library with Helium? I'm going to have a custom board with esp32-c3 board and looking for right library and lora module. Is there any recommendation?

Thanks

Kongduino, 2022-05-24:
Definitely not. As mentioned, among other places, in the Readme, this is a raw LoRa (aka LoRa P2P, or LoRa PHY) library. It does not even do LoRaWAN, let alone other protocols. This is a great library for people who need a lightweight LoRa PHY library.

kargarisaac, 2022-05-25:
> Definitely not. As mentioned, among other places, in the Readme, this is a raw LoRa (aka LoRa P2P, or LoRa PHY) library. It does not even do LoRaWAN, let alone other protocols. This is a great library for people who need a lightweight LoRa PHY library.

Thanks for the reply. What do you suggest to start with? I'm new and looking for a starting point.

morganrallen, 2022-05-25:
> What do you suggest
Don't get involved in 'crypto' ponzi schemes.

closed 2022-05-25 by morganrallen