sandeepmistry/arduino-LoRa #591: Unstable LoRaSimpleGateway.ino

JFPayeur, 2022-08-26:
LoRaSimpleGateway.ino Is unstable out of the box 

I've Just changed this in LoRaSimpleGateway.ino
```
const long frequency = 433E6;  // LoRa Frequency
const int csPin = 5;          // LoRa radio chip select
const int resetPin = 14;        // LoRa radio reset
const int irqPin = 2;          // change for your board; must be a hardware interrupt pin
```

And I'm using a ```ESP32wroom``` and a  ```SX1278 Ra-02 LoRa 433Mhz```

And Bam
```
08:07:19.557 -> Send Message!
08:07:19.932 -> Guru Meditation Error: Core  1 panic'ed (Interrupt wdt timeout on CPU1). 
08:07:20.026 -> 
08:07:20.026 -> Core  1 register dump:
08:07:20.073 -> PC      : 0x4008a638  PS      : 0x00060b35  A0      : 0x8008988a  A1      : 0x3ffbeccc  
08:07:20.166 -> A2      : 0x3ffb8a00  A3      : 0x3ffb8890  A4      : 0x00000004  A5      : 0x00060b23  
08:07:20.260 -> A6      : 0x00060b23  A7      : 0x00000001  A8      : 0x3ffb8890  A9      : 0x00000018  
08:07:20.353 -> A10     : 0x3ffb8890  A11     : 0x00000018  A12     : 0x3ffc186c  A13     : 0x00060b23  
08:07:20.446 -> A14     : 0x007beeb8  A15     : 0x003fffff  SAR     : 0x0000001b  EXCCAUSE: 0x00000006  
08:07:20.540 -> EXCVADDR: 0x00000000  LBEG    : 0x400862c1  LEND    : 0x400862d1  LCOUNT  : 0xfffffffe  
08:07:20.634 -> Core  1 was running in ISR context:
08:07:20.634 -> EPC1    : 0x400db54b  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x00000000
08:07:20.727 -> 
08:07:20.727 -> 
08:07:20.727 -> Backtrace:0x4008a635:0x3ffbeccc |<-CORRUPTED
08:07:20.820 -> 
08:07:20.820 -> 
08:07:20.820 -> Core  0 register dump:
08:07:20.820 -> PC      : 0x4008a7af  PS      : 0x00060035  A0      : 0x800894b3  A1      : 0x3ffbe80c  
08:07:20.915 -> A2      : 0x3ffbeeb8  A3  ⸮
```

Why ? 

Both 
```
  LoRa.onReceive(onReceive);
  LoRa.onTxDone(onTxDone);
```
Are setting onReceive() and onTxDone() function to be called on a hardware interrupt on Pin2  
, This is fine 

But Both onReceive() and onTxDone() functions do Serial.print()

You never whant to do this, 
Functions called from interrupt should do there stuff as quick as possible and end ASAP. 
In this case , the Serial.print() are to long to execute , and the ESP32 panic,
The watchdog expire ````wdt timeout ```` and the ESP32 reboot 

Removing  Serial.print() in onReceive() and onTxDone() functions resolve the issue 

I don't know how to cleanly resolve the issue
But dirty flags in the in onReceive() and onTxDone() functions , and do the Serial.print() in the Main loop worked for me.

IoTThinks, 2022-09-01:
To set flag in omReceive and process in main loop.
Or to create task to process incoming Rx.

Should not do much in an ISR.

Anonymouse-X, 2023-08-25:
I am new to programming and I encounted this one right now. I didnt quite get the " dirty flags in the in onReceive() and onTxDone() functions"

can you send me the code for it? thank you!

juanmartin84, 2024-11-21:
Hello! Sorry for being late, this worked for me:

```
int lastPacketSize; 
void onReceive(int packetSize) {

  lastPacketSize = packetSize;

 }

```
then this function in some place in the main loop (you can do what ever you want here):

```
void loraLoop(){

  if(lastPacketSize != 0){

    Serial.print("Received packet '");

    for (int i = 0; i < lastPacketSize; i++) {
      Serial.print((char)LoRa.read());
    }

    Serial.print("' with RSSI ");
    Serial.println(LoRa.packetRssi());

    lastPacketSize = 0;
    }
  }
```
BR