sandeepmistry/arduino-LoRa #595: what is the latency of a lora 433 mhz ra-02 (SX1278) module

hannescam, 2022-09-30:
i have a project that requiers to send low latency low quantity of data between 100 meter distance

JFPayeur, 2022-09-30:
How much low quantity ?

hannescam, 2022-09-30:
it is to transmit a signal evrey 2 minutes over 100 meters

Kongduino, 2022-09-30:
Latency is not the only problem you'll have, but let's start with that. We'd have to define "low quantity" since what is considered low-quantity data in other fields might be considered as high load for LoRa. So let's consider a typical LoRa/LoRaWAN light payload, 11 bytes. At SF 8, ​BW 500, airtime is 21 ms. This is a fast-speed, low-distance LoRa setting, but 100 meters is nothing, especially at 433 MHz. At SF 12, BW 62.5, airtime would be almost 14 times that, at 288 ms.

Which brings us to the next issue, fair use (which is optional for LoRa, but not LoRaWAN), and beyond just being nice, availability of the band you are using, to ensure reliability of transmission, and reception... While I was typing this, I did a test with some of my devices (I was looking at 868.125, but that besides the point), and received 3 messages that were not from/for me. And that's at night. Imagine daytime.

So depending on the frequency of the transmission, you find find out that LoRa is not a good use case. You might have also to shop for a band that's less noisy than 433.

hannescam, 2022-09-30:
i have done some tests and never recieved a message that was not from me maybe it is because i am living in an erea with not much people like me that experiment with these things and i am planing to use this as a two part timer so one module will send a start signal and the other will recieve it and start the timer untile that side will stop the timer and display it on a 1602 lcd we all love and that means data quantity wont be an issue but if the delay is always consistent i can calculate it out in software and for the unwanted signals it is not hard to simply react on a command like "s" to star the timer so my only concern is that the latency could vary between transmissions and i cant calculate it out

Kongduino, 2022-10-01:
The latency of the LoRa chip will be uniform. The unquantifiable part for now will be based on what MCU you will use with it, on both sides, and how busy they are besides sending / receiving the signal. Especially on the receiving side – how long it will take to wake up / react to the DIO0 interrupt, etc.

hannescam, 2022-10-01:
the mcos are both atmega88 with this biitloader: [https://github.com/MCUdude/MiniCore](https://github.com/MCUdude/MiniCore) because all atmega328 are out of stock and the arduino boards were too expensive and the recieving end will only wait for that package

Kongduino, 2022-10-01:
An ATmega88 has 8 KB of Flash and 1 KB of RAM: the simplest example code (Sender) will eat up 4 KB of Flash and 1/3 to half of the RAM, you will indeed not do much else than send and receive signals. But I supposed iy should be good enough to send a short packet, and receive it – and possibly turn on/off a relay of some sort. Should you run out of memory, paring down the LoRa library could also be an option – there's stuff in there that's not critical.

hannescam, 2022-10-01:
the reciever has also to manage a hc-sr04 (ultrasonic module) and a 1602 and the transmitter has only an led and an hc-sr04

hannescam, 2022-10-01:
and do you know where to get these ra-02 from mouser (please post a link)

Kongduino, 2022-10-01:
I don't think Mouser carries them. But then again I wouldn't know for sure as Ra-01/Ra-02 chips are not very good, and I tried to stay away from them.

hannescam, 2022-10-01:
oh ok and which modules do you recomend then (please provide a mouser link if possible) 
thankyou

JFPayeur, 2022-10-01:
> I don't think Mouser carries them. But then again I wouldn't know for sure as Ra-01/Ra-02 chips are not very good, and I tried to stay away from them.

Im using ra-02 chip and I achieve real world usage of 2-3km. What problem do you had with these? (sorry to hijack the tread)

hannescam, 2022-10-01:
i agree but i need them from mouser

Kongduino, 2022-10-02:
2-3 km is pretty good for Ra-02, but it's pretty crappy for SX127x. With most of my modules, I got up to 10.x km, at either 433, 868 and 915.

As for Mouser, I wouldn't know, as I never use it – I get most of my stuff from China.

hannescam, 2022-10-02:
ok i will order my ra-02 from amazon because on mouser i cant find anything
i will close this issue because i think i have evrey info i need

closed 2022-10-02 by hannescam