sandeepmistry/arduino-LoRa #600: Correct the frequency error corrupts the data

StevenArduino1, 2022-10-16:
I have a SX1276 chip and I am successfully sending 100 bytes of data at 915E6 frequency (Australia).  I am sending data back and forth ( e.g. send data and then receive a OK handshake by the sender unit from the receiver that data was received) and I am recording the frequency error on both sides.  I noticed that at the sender unit when it gets a handshake reply the frequency error is about -15000 and the frequency error at the receiver is about + 15000. This makes sense that if one is recording a positive shift than the other should be a negative shift.  In perusing to makes things optimal, at the sender unit I increased the frequency from 915E6 to 915015000  (via LoRa.begin(915015000);)  .  On the next reading the frequency error was only about 200/-200 both on the sender and receiver, fabulous, the frequencies are aligned. However I was getting lots of data corruption of the wrong numbers projected on the receiver end.  I then changed the sender frequency back to 915E6 and the frequency error went back to -15000 but then the data was perfect again.  The SNR is around 10 in all tests. I am testing the system with the sender and receiver a few rooms away in the house.  The RSSI is about -80.  Can anyone explain what is going on?

Kongduino, 2022-10-17:
These are center frequencies, and must match on all devices. Depending on the bandwidth and spreading factor you'll see (if you have an SDR have a look) how the transmission spreads around the center frequency. That is normal. Just play around with BW and SF (the higher the SF, the lower the BW, and the stronger and slower the signal will be).

StevenArduino1, 2022-10-24:
Kongduino, Thank you for your response. I am still confused by as you said "enter frequencies, and must match on all devices".  When I matched the centre frequencies by adjusting the frequency of the sender unit I received many errors in the data. I should have not received errors because I matched the middle frequencies. I then changed it back to 915E6 and no problems. Did anyone else get such problems when they adjusted the frequencies to minimise the frequency error?

Kongduino, 2022-10-24:
Hi. When I said
> These are center frequencies, and must match on all devices.

It means that:

* both devices must be set at the same frequency, eg 915e6;
* the frequency you set is the center frequency, ie, depending on the bandwidth you set, it will send the chirps on a range, with the frequency halfway between both minimum and maximum frequency. For example, if you set BW to 250 KHz, and the frequency to 915e6, the chirps will be sent between 914.875 MHz and 915.125 MHz.

[image]

So if you set one of the devices at a frequency that's within the bandwidth, but not on the same center frequency, the chirps will be decoded wrong, and you'll get errors. In short: don't do that. :-)