sandeepmistry/arduino-LoRa #616: Registering Device on The Things Network

DeveloperMarcial, 2022-12-02:
**What LoRaWAN version is this library implementing?**
* LoRaWAN Specification 1.0.0,
* LoRaWAN Specification 1.0.1,
* LoRaWAN Specification 1.0.2,
* LoRaWAN Specification 1.0.3,
* LoRaWAN Specification 1.0.4, or
* LoRaWAN Specification 1.1.0?

**What Regional Parameters version is this library implementing?**
* TS001 Technical Specification 1.0.0,
* TS001 Technical Specification 1.0.1,
* RP001 Technical Specification 1.0.2,
* RP001 Technical Specification 1.0.2 revision B,
* RP001 Technical Specification 1.0.3 revision A,
* RP001 Technical Specification 1.1 revision A,
* RP001 Technical Specification 1.1 revision B,
* RP002 Technical Specification 1.0.0,
* RP002 Technical Specification 1.0.1,
* RP002 Technical Specification 1.0.2, or
* RP002 Technical Specification 1.0.3?

Kongduino, 2022-12-03:
It is **_NOT_** implementing LoRaWAN – hence the name LoRa, not LoRaWAN... It's even mentioned in the README.

DeveloperMarcial, 2022-12-03:
I am a noob with LoRa but now the WAN in LoRaWAN has crystallized for me. Thankyou.

FAQ #4 makes sense to me now. (24-hours ago those words did not.)

closed 2022-12-03 by DeveloperMarcial