sandeepmistry/arduino-LoRa #624: esp32 lora endpacket stuck

matg41, 2022-12-31:
Hello.
I am using a lora-ra2 sz1278 433 MHz and ESP32 to read the state of a GPIO (27) and send to a gateway. When I put the GPIO in input pull-up, the program stuck in `lora.endpacket()`. Also when I put it on input mode and connect it to gnd on ESP, I get the same problem.
I solved it by using a 5v power supply and connect the GPIO to gnd and vcc of it. The problem is solved.
However, I did not manage to understand the problem.

Thanks?

Kongduino, 2023-01-01:
What's `GPIO27` connected to? If it's a sensor or other similar device, it needs to share a ground line with the ESP32.

matg41, 2023-01-01:
> What's `GPIO27` connected to? If it's a sensor or other similar device, it needs to share a ground line with the ESP32.

It is connected to the emitter of a transistor which is activated by a 12V signal from a driver. The collector is connected to the 5V power supply. The `GPIO27` is also connected (with a 5KOhm resistor) to the GND of 5V power supply. I use the same power supply to turn on my ESP32.