sandeepmistry/arduino-LoRa #627: Problem with ESP32 and LoRa.h Core 1 panic'ed

pyrotechnik, 2023-01-25:
I have problem with any example from LoRa.h.
My Code:
```
#include <SPI.h>              // include libraries
#include <LoRa.h>

#define  LoRa_SCK   14
#define  LoRa_MISO  12
#define  LoRa_MOSI  13
#define  LoRa_SS    15
#define  LoRa_RST   32    //  LoRa_Reset
#define  LoRa_DI0        33    //  LoRa_IRQ
#define  BAND    433E6        // change for your board; must be a hardware interrupt pin

String message = "";

void PrintReceivedMessage(void * parameter){
  Serial.print("Gateway Receive: ");
  Serial.println(message);      
  vTaskDelete(NULL);
}

void setup() {
  Serial.begin(115200);                   // initialize serial
  while (!Serial);

  pinMode(LoRa_DI0, INPUT);

  SPIClass SPI_LORA(HSPI);
  SPI_LORA.begin(LoRa_SCK, LoRa_MISO, LoRa_MOSI, LoRa_SS);

  LoRa.setSPI(SPI_LORA);
  LoRa.setPins(LoRa_SS, LoRa_RST, LoRa_DI0);

  if (!LoRa.begin(433E6)) {
    Serial.println("LoRa init failed. Check your connections.");
    while (true);                       // if failed, do nothing
  }
  LoRa.setSyncWord(0x39);

  Serial.println("LoRa init succeeded.");

    LoRa.onReceive(onReceive);
  LoRa.onTxDone(onTxDone);
  LoRa_rxMode();

}

void loop() {
  if (runEvery(5000)) { // repeat every 5000 millis

    String message = "HeLoRa World! ";
    message += "I'm a Gateway! ";
    message += millis();

    LoRa_sendMessage(message); // send a message

 
    Serial.println("Send Message!");
  }
}

void LoRa_rxMode(){
  LoRa.disableInvertIQ();               // normal mode
  LoRa.receive();                       // set receive mode
}

void LoRa_txMode(){
  LoRa.idle();                          // set standby mode
  LoRa.enableInvertIQ();                // active invert I and Q signals
}

void LoRa_sendMessage(String message) {
  LoRa_txMode();                        // set tx mode
  LoRa.beginPacket();                   // start packet
  LoRa.print(message);                  // add payload
  LoRa.endPacket(true);                 // finish packet and send it
}

void onReceive(int packetSize) { 
  
  message = "";
  
  while (LoRa.available()) {
    message += (char)LoRa.read();
  }
  
  xTaskCreate(
       PrintReceivedMessage,
       "PrintReceivedMessage Function",
       10000,             // Stack size (bytes)
       NULL,             // Parameter
       5,                // Task priority
       NULL              // Task handle
   );

}

void onTxDone() {
  LoRa_rxMode();
}

boolean runEvery(unsigned long interval){
  
  static unsigned long previousMillis = 0;
  unsigned long currentMillis = millis();
  if (currentMillis - previousMillis >= interval)
  {
    previousMillis = currentMillis;
    return true;
  }
  return false;
  
}
```

Even if I try a simple receiver, the result is always the same.

```
#include <SPI.h>              // include libraries
#include <LoRa.h>

#define  LoRa_SCK   14
#define  LoRa_MISO  12
#define  LoRa_MOSI  13
#define  LoRa_SS    15
#define  LoRa_RST   32    //  LoRa_Reset
#define  LoRa_DI0        33    //  LoRa_IRQ
#define  BAND    433E6        // change for your board; must be a hardware interrupt pin

void setup() {
  Serial.begin(115200);                   // initialize serial
  while (!Serial);
  SPIClass SPI_LORA(HSPI);
  SPI_LORA.begin(LoRa_SCK, LoRa_MISO, LoRa_MOSI, LoRa_SS);

  LoRa.setSPI(SPI_LORA);
  LoRa.setPins(LoRa_SS, LoRa_RST, LoRa_DI0);

  if (!LoRa.begin(433E6)) {
    Serial.println("LoRa init failed. Check your connections.");
    while (true);                       // if failed, do nothing
  }
  LoRa.setSyncWord(0x39);

  Serial.println("LoRa init succeeded.");

}

void loop() {
  // try to parse packet
  int packetSize = LoRa.parsePacket();
  if (packetSize) {
    // received a packet
    Serial.print("Received packet '");

    // read packet
    while (LoRa.available()) {
      String LoRaData = LoRa.readString();
      Serial.print(LoRaData); 
    }

    // print RSSI of packet
    Serial.print("' with RSSI ");
    Serial.println(LoRa.packetRssi());
  }
}
```

The error code is as follows:

```
LoRa init succeeded.
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x400ea2d5  PS      : 0x00060530  A0      : 0x800d0f9a  A1      : 0x3ffb1ea0  
A2      : 0x0021002f  A3      : 0x3ffb8544  A4      : 0x0000002c  A5      : 0x3ffb8574  
A6      : 0x00000020  A7      : 0x00000002  A8      : 0x00000000  A9      : 0x00000000  
A10     : 0x3ffb8544  A11     : 0x00000000  A12     : 0x00000030  A13     : 0x3ffbbca8  
A14     : 0x00000000  A15     : 0x00000000  SAR     : 0x00000011  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x0021002f  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0x00000000  

ELF file SHA256: 0000000000000000

Backtrace: 0x400ea2d5:0x3ffb1ea0 0x400d0f97:0x3ffb1ec0 0x400d10f0:0x3ffb1ef0 0x400d1240:0x3ffb1f10 0x400d0d49:0x3ffb1f30 0x400d0d57:0x3ffb1f50 0x400d0dea:0x3ffb1f70 0x400d2f9d:0x3ffb1fb0 0x400861e1:0x3ffb1fd0
```

I don't know where to look for the error. I will be glad for any idea.
Thank you

Kongduino, 2023-01-26:
The first thing to do is to stick the error text in [the decoder](https://github.com/me-no-dev/EspExceptionDecoder).
[image]
This will tell you, more or less, what it causing the error.

pyrotechnik, 2023-01-26:
```
0x400d18d7: spiTransferByteNL at /Users/davidjezek/Library/Arduino15/packages/esp32/hardware/esp32/1.0.6/cores/esp32/esp32-hal-spi.c line 782
0x400d0e9e: SPIClass::transfer(unsigned char) at /Users/davidjezek/Library/Arduino15/packages/esp32/hardware/esp32/1.0.6/libraries/SPI/src/SPI.cpp line 158
0x400d0f27: LoRaClass::singleTransfer(unsigned char, unsigned char) at /Users/davidjezek/Documents/Arduino/libraries/src/LoRa.cpp line 740
0x400d0f54: LoRaClass::readRegister(unsigned char) at /Users/davidjezek/Documents/Arduino/libraries/src/LoRa.cpp line 725
0x400d1158: LoRaClass::parsePacket(int) at /Users/davidjezek/Documents/Arduino/libraries/src/LoRa.cpp line 222
0x400d0cde: loop() at /Users/davidjezek/Documents/Arduino/lora_test_2/lora_test_2.ino line 34
0x400d21e1: loopTask(void*) at /Users/davidjezek/Library/Arduino15/packages/esp32/hardware/esp32/1.0.6/cores/esp32/main.cpp line 23
0x40086155: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c line 143
```

Kongduino, 2023-01-28:
So the problem start at line 34 of your code:
`int packetSize = LoRa.parsePacket();`

Next step, [line 222 of the lib](https://github.com/sandeepmistry/arduino-LoRa/blob/master/src/LoRa.cpp#L222).
`int irqFlags = readRegister(REG_IRQ_FLAGS);`

Then [line 725](https://github.com/sandeepmistry/arduino-LoRa/blob/master/src/LoRa.cpp#L725).
`  return singleTransfer(address & 0x7f, 0x00);`

Then [line 740](https://github.com/sandeepmistry/arduino-LoRa/blob/master/src/LoRa.cpp#L740).
`_spi->transfer(address);`

Onwards to the Espressif BSP, file `SPI.cpp`. This is a little more complicated because you are using a very old BSP, 1.06. But `SPIClass::transfer` calls `spiTransferByteNL` in `esp32-hal-spi.c`. You'll have to check what line 782 is for you – `uint8_t spiTransferByteNL(spi_t * spi, uint8_t data)` for me starts at line 1174...

But that's the hill where it's dying. An SPI issue. Could it be that it doesn't like you to play with `HSPI`? Anyway that's as much as I can infer from the code and logs.

amstaff8, 2023-04-20:
Hello

I had a similar problem. I was trying to build a simple sender and got a panic error. I solved the problem by putting the SPI object construction outside of the setup function. I think in that way it doesn't go outside of scope. I hope that it will help you.

`
#include <SPI.h>
// - Lora Library: https://github.com/sandeepmistry/arduino-LoRa
#include <LoRa.h>

//SPI pin
#define  LoRa_SCK   36
#define  LoRa_MISO  37
#define  LoRa_MOSI  38
#define  LoRa_SS    2
#define  LoRa_RST   1    //  LoRa_Reset
#define  LoRa_DI0   35    //  LoRa_IRQ

//initialise HSPI
SPIClass hspi(HSPI);

void setup() {

  hspi.begin(LoRa_SCK, LoRa_MISO, LoRa_MOSI, LoRa_SS);

  LoRa.setSPI(hspi);
  LoRa.setPins(LoRa_SS, LoRa_RST, LoRa_DI0);

  Serial.begin(9600);

  Serial.println("LoRa Sender");

  /*and 1 -> Freq. 868.0-868.6 MHz | Duty cycle 1% | Max ERP 25 mW / 14 dBm
  Band 2 -> Freq. 868.7-869.2 MHz | Duty cycle 0.1% | Max ERP 25 mW / 14 dBm
  Band 3 -> Freq. 869.4-869.65 MHz | Duty cycle 10% | Max ERP 500 mW / 27 dBm
  Band 4 -> Freq. 869.7-870 MHz | Duty cycle 1% | Max ERP 25 mW / 14 dBm

  Source : ETSI TR 103 526, Table 12, Page 35
  
  
  by default:

  enables explicit header mode (default)
  preamble lenght is 8 by default
  coding rate is 4/5 by default
  spreading factor is 7 by default
  bandwidth is 125kHz by default 
  CRC is not used by default */

  LoRa.setPins(LoRa_SS, LoRa_RST, LoRa_DI0);

  if (!LoRa.begin(868E6)) {
    Serial.println(F("Starting LoRa failed!"));
    while (1);
  }

  LoRa.setTxPower(12);  //12 dbm default on PA_OUTPUT_PA_BOOST_PIN

  LoRa.setSpreadingFactor(7); //spreading factor 7

  LoRa.setSignalBandwidth(125E3); //bandwidth 125 kHz

  Wire.begin(SDA_PIN, SCL_PIN);
  //my other code...
}

//my other code...
`

colarocker, 2023-11-15:
The SPI initialization described by @amstaff8 didn't help in my case, still getting panics :/

phutai9101, 2024-03-05:
hi , I am doing a project about esp32 with module lora sx1278. I can't start the lora module and get an error. Help meeee! thanks for reading.

#include <SPI.h>
#include <LoRa.h>

#define SS_PIN 5       // Chân SS (Slave Select) kết nối với DIO5 của SX1278
#define RST_PIN 2     // Chân RESET kết nối với DIO2 của SX1278
#define DI0_PIN 4     // Chân DIO0 kết nối với DIO4 của SX1278
#define LED_PIN 15     // Chân GPIO 15 kết nối với LED

#define BAND 433E6     // Tần số truyền thông (433MHz)

void setup() {
  pinMode(SS_PIN,OUTPUT);
  digitalWrite(SS_PIN,HIGH);

  LoRa.begin(433E6);
  // Khởi tạo module LoRa
  LoRa.setPins(SS_PIN, RST_PIN, DI0_PIN);
 // LoRa.setSPIFrequency(433E6);
  Serial.begin(9600);

  if (!LoRa.begin(BAND)) {
    Serial.println("Khởi tạo LoRa thất bại. Kiểm tra kết nối chân.");
    while (true);
  }
  
  pinMode(LED_PIN, OUTPUT);  // Thiết lập chân LED là OUTPUT
}

void loop() {
  if (LoRa.parsePacket()) {  // Kiểm tra xem có gói dữ liệu đang được nhận hay không
    while (LoRa.available()) {  // Đọc dữ liệu trong gói
     String data = LoRa.readString();
      Serial.println("Đã nhận dữ liệu: " + data);
      if(data == "bat"){
        Serial.println("bat den");
        digitalWrite(LED_PIN, HIGH); 
        delay(100); // Bật LED
        digitalWrite(LED_PIN,LOW);
      }
      else if(data == "tat"){
        Serial.println("tắt đèn");
        digitalWrite(LED_PIN,LOW);
      }
  }
}
}
Erorr:
[image]