sandeepmistry/arduino-LoRa #630: Starting LoRa failed

serhatbozasb, 2023-02-17:
My lora is this

[image]
 
unfortunately i can't run lora.
[image]
[image]
[image]

Can you help me?

serhatbozasb, 2023-02-17:
I made my connections according to this template.
[image]

Kongduino, 2023-02-17:
You provided photos and a connection diagram, which is nice. So many people forget lol. But the code would be nice too!

serhatbozasb, 2023-02-18:
I am using the original code. I replace 915 with 868

[image]
[image]