sandeepmistry/arduino-LoRa #632: Problem with aes256

alemori, 2023-02-20:
I am encrypting data that I send by Lora with Aes256, I use:#include "mbedtls/aes.h" #include <mbedtls/md.h>, and when I receive it I get a memory dump.

#include <SPI.h>
#include <LoRa.h>

#ifdef ARDUINO_SAMD_MKRWAN1300
#error "This example is not compatible with the Arduino MKR WAN 1300 board!"
#endif
#define ss 5
#define rst 2
#define dio0 4
void setup() {
  Serial.begin(9600);
  while (!Serial);

  Serial.println("LoRa Receiver Callback");
  LoRa.setPins(ss, rst,dio0);
  
    if (!LoRa.begin(433E6)) {
      Serial.println("Starting LoRa failed!");
      while (1);
    }
  

  // Uncomment the next line to disable the default AGC and set LNA gain, values between 1 - 6 are supported
  // LoRa.setGain(6);
  
  // register the receive callback
  LoRa.onReceive(onReceive);

  // put the radio into receive mode
  LoRa.receive();
}

void loop() {
  // do nothing
}

void onReceive(int packetSize) {
  // received a packet
  Serial.print("Received packet '");

  // read packet
  for (int i = 0; i < packetSize; i++) {
    Serial.print((char)LoRa.read());
  }

  // print RSSI of packet
  Serial.print("' with RSSI ");
  Serial.println(LoRa.packetRssi());
}

⸮E0⸮⸮1⸮⸮ʄ⸮⸮⸮⸮⸮J⸮LoRa Receiver Callback
Guru Meditation Error: Core  1 panic'ed (Interrupt wdt timeout on CPU1). 

Core  1 register dump:
PC      : 0x4008aa52  PS      : 0x00060b35  A0      : 0x800899c6  A1      : 0x3ffbf13c  
A2      : 0x3ffb8a00  A3      : 0x3ffb8890  A4      : 0x00000004  A5      : 0x00060b23  
A6      : 0x00060b23  A7      : 0x00000001  A8      : 0x3ffb8890  A9      : 0x00000018  
A10     : 0x3ffb8890  A11     : 0x00000018  A12     : 0x3ffc2adc  A13     : 0x00060b23  
A14     : 0x007bf328  A15     : 0x003fffff  SAR     : 0x0000001b  EXCCAUSE: 0x00000006  
EXCVADDR: 0x00000000  LBEG    : 0x4008638d  LEND    : 0x4008639d  LCOUNT  : 0xfffffffc  
Core  1 was running in ISR context:
EPC1    : 0x400db87b  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x00000000

Backtrace: 0x4008aa4f:0x3ffbf13c |<-CORRUPTED

Core  0 register dump:
PC      : 0x4008abeb  PS      : 0x00060035  A0      : 0x800895ef  A1      : 0x3ffbea4c  
A2      : 0x3ffbf328  A3 ⸮

Decoding 3 results
0x4008aa52: vListInsert at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/freertos/list.c line 166 (discriminator 1)
0x4008638d: strlen at /builds/idf/crosstool-NG/.build/HOST-i686-w64-mingw32/xtensa-esp32-elf/src/newlib/newlib/libc/machine/xtensa/strlen.S line 84
0x4008639d: strlen at /builds/idf/crosstool-NG/.build/HOST-i686-w64-mingw32/xtensa-esp32-elf/src/newlib/newlib/libc/machine/xtensa/strlen.S line 96

Kongduino, 2023-02-21:
Please surround your code with 3 backticks + c and 3 backticks to make it readable. Or select it and click add code...

Kongduino, 2023-02-22:
The first thing to do is to put the error log into the [ESP Exception Decoder](https://github.com/me-no-dev/EspExceptionDecoder). This will tell you – hopefully – where the problem happens.

After that, there are a few general issues with your code.

• Do not use Arduino `String` objects. They're not only wasteful, and slow down your program, but also, since your AES functions and the LoRa library use char arrays, there's no reason not to use that throughout the code.

• Also, you're sending a hex string instead of just the bytes, which is again wasteful and doesn't make sense. You could just send the char array with `LoRa.write(buf, len)`. A much shorter and faster transmission...

• You're sending the key, in clear, in the packet. WHY!?!? You should only send the IV (first) + ciphertext blocks.

• You're using a static IV. DON'T. It defeats the purpose of an IV. Generate a non-predictable IV on the spot, every time. You could use the random generator that the SX1276 provides, except it is broken in this library. See [LoRandom](https://github.com/Kongduino/LoRandom) for an example of a functioning LoRa RNG.

alemori, 2023-02-22:
Thank you for your response. I activated the debug and the output is in the first code I sent, which is about the reception of the package. The truth is that I use String because it is easier for me, and it is what I saw in examples, but I will try to modify the code. On the other hand, the dump will stop when I avoid the use of the onRecive() function. I would like to know what is the problem with the dump. Please wait for instructions, and I clarify that I do not have too much expertise in the handling of c++ but with a guide I can follow the problem. Thank you very much for your help.

⸮E0��⸮⸮1⸮⸮ʄ⸮�⸮⸮⸮�⸮J⸮LoRa Receiver Callback
Guru Meditation Error: Core 1 panic'ed (Interrupt wdt timeout on CPU1).

Core 1 register dump:
PC : 0x4008aa52 PS : 0x00060b35 A0 : 0x800899c6 A1 : 0x3ffbf13c
A2 : 0x3ffb8a00 A3 : 0x3ffb8890 A4 : 0x00000004 A5 : 0x00060b23
A6 : 0x00060b23 A7 : 0x00000001 A8 : 0x3ffb8890 A9 : 0x00000018
A10 : 0x3ffb8890 A11 : 0x00000018 A12 : 0x3ffc2adc A13 : 0x00060b23
A14 : 0x007bf328 A15 : 0x003fffff SAR : 0x0000001b EXCCAUSE: 0x00000006
EXCVADDR: 0x00000000 LBEG : 0x4008638d LEND : 0x4008639d LCOUNT : 0xfffffffc
Core 1 was running in ISR context:
EPC1 : 0x400db87b EPC2 : 0x00000000 EPC3 : 0x00000000 EPC4 : 0x00000000

Backtrace: 0x4008aa4f:0x3ffbf13c |<-CORRUPTED

Core 0 register dump:
PC : 0x4008abeb PS : 0x00060035 A0 : 0x800895ef A1 : 0x3ffbea4c
A2 : 0x3ffbf328 A3 ⸮

Decoding 3 results
0x4008aa52: vListInsert at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/freertos/list.c line 166 (discriminator 1)
0x4008638d: strlen at /builds/idf/crosstool-NG/.build/HOST-i686-w64-mingw32/xtensa-esp32-elf/src/newlib/newlib/libc/machine/xtensa/strlen.S line 84
0x4008639d: strlen at /builds/idf/crosstool-NG/.build/HOST-i686-w64-mingw32/xtensa-esp32-elf/src/newlib/newlib/libc/machine/xtensa/strlen.S line 96

Kongduino, 2023-02-22:
The debug log isn't at its clearest but it does seem to point to a string issue.m Unfortunately it doesn't say where it originates from in your code, just where in the IDF.

Also, another mistake I hadn't spotted before, silly me. Delays between LoRa sends should be in tens of seconds, not milliseconds. If you do `delay(100);` there's not enough time to send your packet: even at SF7 BW125, you're sending packets of around bytes, which takes more than 100 ms. At SF10 BW125, we'd talk about 650 ms. And that's just the technical reason. You also should not flood the airwaves with your stuff – there may be other people around you trying to send / receive.

And on the subject of delays, do not use `delay` – delays are blocking. In `loop()` you should use a reference time, acquired from `millis()`, and compare with the present time. eg

```c
// global object
uint32_t t0;
// time between pings in milliseconds
#define PING_DELAY 10000 // ten seconds

void setup() {
  // last line in setup()
  t0 = millis(); // get an initial time reference
}

void loop() {
  if(millis() - t0 > MY_PING_DELAY) {
    // send a lora packet
    [...]
    t0 = millis(); // update time reference
  }
}
```
That way your code is non-blocking.

morganrallen, 2023-02-23:
@alemori does it seem to pause for a moment between `LoRa Receiver Callback` and `Guru Meditation Error: Core 1 panic'ed (Interrupt wdt timeout on CPU1).` 

That error message indicates something it holding the task and taking too long. Check any loops in the decryption portion of the code, something could be looping forever.

alemori, 2023-02-23:
@alemori does it seem to pause for a moment between LoRa Receiver Callback and Guru Meditation Error: Core 1 panic'ed (Interrupt wdt timeout on CPU1). ?

That error message indicates something it holding the task and taking too long. Check any loops in the decryption portion of the code, something could be looping forever.

alemori, 2023-02-23:
Thanks again. I think the problem may come from the timing issue, I did not take into account not to use delay, but to produce the delay with millis(). I will make the changes and see what result I get. As for the issue of avoiding String, I understand that they can be more inefficient, but they should not cause a dump.

Kongduino, 2023-02-23:
Considering that the error happens in `strlen.S` and `list.c` I wouldn't be so quick dismissing the usage of `String` as an issue...

alemori, 2023-02-23:
Hi, I made the proposed changes regarding the delays and changing the String. It still gives a memory overwrite. Copy the send and receive sketches. Thank you very much for your help.

//   Sketch Emisor

#include <SPI.h>
#include <LoRa.h>
#include <Wire.h>
#include "DHT.h"
#include "WiFi.h"

#include <SoftwareSerial.h>

/*
   Clase para registrar nodo
*/

#include <ArduinoJson.hpp>
#include <ArduinoJson.h>
/*
   Biblioteca AES256
*/
#include "mbedtls/aes.h"
#include <mbedtls/md.h>

#define ss 5
#define rst 2
#define dio0 4

#if !defined(CONFIG_BT_ENABLED) || !defined(CONFIG_BLUEDROID_ENABLED)
#error Bluetooth is not enabled! Please run `make menuconfig` to and enable it
#endif
#define DHTPIN 33     // Pin donde está conectado el sensor
#define DHTTYPE DHT22   // Sensor DHT22

DHT dht(DHTPIN, DHTTYPE);

String json;
String json2;

int counter = 0;
int counterWakeUp = 0;
String message = "";
char incomingChar;
int s1 = 0;
int s2 = 0;
int s3 = 0;
int mando = 0;
int alerta_a = 1;
float primer_ciclo = 0;
long tiempo_inicial = 0;

StaticJsonDocument<200> doc;
StaticJsonDocument<50> doc2;

#define uS_TO_S_FACTOR 1000000ULL  /* Conversion factor for micro seconds to seconds */
#define TIME_TO_SLEEP  9        /* Time ESP32 will go to sleep (in seconds) */
uint32_t t0;
#define MY_PING_DELAY 5000 // ten seconds

void print_wakeup_reason() {
  esp_sleep_wakeup_cause_t wakeup_reason;
  wakeup_reason = esp_sleep_get_wakeup_cause();

  switch (wakeup_reason)
  {
    case ESP_SLEEP_WAKEUP_EXT0 : Serial.println("Wakeup caused by external signal using RTC_IO"); break;
    case ESP_SLEEP_WAKEUP_EXT1 : Serial.println("Wakeup caused by external signal using RTC_CNTL"); break;
    case ESP_SLEEP_WAKEUP_TIMER : Serial.println("Wakeup caused by timer"); break;
    case ESP_SLEEP_WAKEUP_TOUCHPAD : Serial.println("Wakeup caused by touchpad"); break;
    case ESP_SLEEP_WAKEUP_ULP : Serial.println("Wakeup caused by ULP program"); break;
[... 348 lines cut ...]
  pos1 = mynewcopystring.indexOf("&");
  mynewcopystring = mynewcopystring.substring(0, pos1);
  return (mynewcopystring);
}

// Error y decodificacion

Guru Meditation Error: Core  1 panic'ed (Interrupt wdt timeout on CPU1). 

Core  1 register dump:
PC      : 0x4008e822  PS      : 0x00060235  A0      : 0x8008d4c6  A1      : 0x3ffbf3dc  
A2      : 0x3ffb8a00  A3      : 0x3ffb8890  A4      : 0x00000004  A5      : 0x00060223  
A6      : 0x00060223  A7      : 0x00000001  A8      : 0x3ffb8890  A9      : 0x00000018  
A10     : 0x3ffb8890  A11     : 0x00000018  A12     : 0x3ffc5a4c  A13     : 0x00060223  
A14     : 0x007bf5d8  A15     : 0x003fffff  SAR     : 0x0000000a  EXCCAUSE: 0x00000006  
EXCVADDR: 0x00000000  LBEG    : 0x40089939  LEND    : 0x40089949  LCOUNT  : 0xfffffffb  
Core  1 was running in ISR context:
EPC1    : 0x400efa8b  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x00000000

Backtrace: 0x4008e81f:0x3ffbf3dc |<-CORRUPTED

0x4008e822: vListInsert at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/freertos/list.c line 183
0x40089939: strlen at /builds/idf/crosstool-NG/.build/HOST-i686-w64-mingw32/xtensa-esp32-elf/src/newlib/newlib/libc/machine/xtensa/strlen.S line 84
0x40089949: strlen at /builds/idf/crosstool-NG/.build/HOST-i686-w64-mingw32/xtensa-esp32-elf/src/newlib/newlib/libc/machine/xtensa/strlen.S line 96
0x400efa8b: uart_hal_write_txfifo at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/hal/uart_hal_iram.c line 35

Kongduino, 2023-02-24:
Your sketch is again unreadable, please edit the last post and put 3 backticks each before and after. One comment I have though – besides the fact that there are `String` objects all over the code still – is that your key isn't 16 bytes in length )or 24 or 32, depending on which AES you were thinking of using). That in itself is a problem. Whether it's THE problem, no idea, but you'd better fix that too.
Also, the length of the plaintext to encrypt has to be a multiple of 16.

alemori, 2023-02-24:
Thank you very much for your comments, I will try to clean up the code to make it readable, and correct the points you indicate. Thanks for your patience and help.

alemori, 2023-02-28:
I have to modify the code, and look at the encryption issue, maybe changing the library or method. Also look at the locking issue. For the moment I close the case, and thank you very much for all the information you gave me.

closed 2023-02-28 by alemori