sandeepmistry/arduino-LoRa #656: lorasender example stops working and no signal received

EnchantedWindow, 2023-07-23:
I'm using two 868 mhz rfm95 modules, and have loaded up an Arduino Nano and an Uno to run LoraSender and LoRaReceiver. I've used an 8 channel logic level converter for both and wired it up according to the instructions, I haven't used DI0 since I do not need receiver callback. I've only changed the parts of the code so I can use my 868 mhz modules. The serial monitor for LoraSender shows
LoRa Sender
Sending packet: 0
Sending packet: 1
Then it stops and shows nothing else instead of looping and going on increasing the counter value. The receiver only prints 'LoRa Receiver' and doesn't receive any signal. And yes ofc I've used both systems at the same time. I have not supplied an external 3.3v supply to the modules.

morganrallen, 2023-07-24:
Impossible to say without seeing your code. Shot in the dark guess... you're sending again before the previous packet is finished.

EnchantedWindow, 2023-07-24:
> Impossible to say without seeing your code. Shot in the dark guess... you're sending again before the previous packet is finished.

I changed the LoraSender code on line 12 from
  "if (!LoRa.begin(915E6)) {"
  to
  "  if (!LoRa.begin(868E6)) {"
and changed line 10 on LoraReceiver from
" if (!LoRa.begin(915E6)) {"
to
" if (!LoRa.begin(868E6)) {"
I'm not sure about sending the next packet before the previous one is done since I'm just viewing the serial monitor.