sandeepmistry/arduino-LoRa #657: Receiver nt receiving from all senders

arifainchtein, 2023-07-24:
Hello,
I have made custom boards using Esp32 WROOM 32E and Ra-02 as lora ic.  
I have two senders S1 and S2, which are deivces will collect sensor data and send it to the receiver R1.
The issue is that R1 is not reliably receiving data from S1 and S2.  Some times I restart R1 and it will receive from S1, others i restart R1 and it will start receiving from S2 but not S1, occasionally it receives from both.

 I have also check to see if the receiver need to be on before the senders and that did not make any difference.
Here is the code that i use to initialise, it is the same in both the senders and receiver:
<code>
 SPI.begin(SCK,MISO,MOSI);
pinMode(LoRa_SS, OUTPUT);
pinMode(LORA_RESET, OUTPUT);
pinMode(LORA_DI0, INPUT);
digitalWrite(LoRa_SS, HIGH);
delay(100);
LoRa.setPins(LoRa_SS,LORA_RESET,LORA_DI0);

if (!LoRa.begin(433E6)) {
    Serial.println("Starting LoRa failed!");
    while (1)
      ;
   } else {
    Serial.println("Starting LoRa worked!");
    loraActive = true;
  }
</code>

Here is the basic code of the sender

<code>
void LoRa_txMode() {
  LoRa.idle();             // set standby mode
  LoRa.disableInvertIQ();  // normal mode
}
void LoRa_rxMode() {
  LoRa.disableInvertIQ();  // normal mode
  LoRa.receive();          // set receive mode
}
void sendMessage() {
   LoRa_txMode();     
  LoRa.beginPacket();  // start packet
  LoRa.write((uint8_t*)&panchoTankFlowData, sizeof(PanchoTankFlowData));
  LoRa.endPacket(true);  // finish packet and send it
  LoRa_rxMode(); 
  msgCount++;        /
}
</code>

and here is the receiving code:
<pre>
void onReceive(int packetSize) {
    if (packetSize == 0) return;         
    Serial.print("received lora size=");
    Serial.println( packetSize);
    int hour = currentTimerRecord.hour;
    int minute = currentTimerRecord.minute;
    int displaytime = (hour * 100) + minute;
    if(packetSize==sizeof(GloriaTankFlowPumpData)){
        LoRa.readBytes((uint8_t*)&gloriaTankFlowPumpData,sizeof(GloriaTankFlowPumpData));
        Serial.print("GloriaTankFlow from ");
        Serial.println(gloriaTankFlowPumpData.devicename);
        gloriaTankFlowPumpData.rssi = LoRa.packetRssi();
        gloriaTankFlowPumpData.snr = LoRa.packetSnr();
        Serial.print("GloriaTankFlow rssi ");
        Serial.println(gloriaTankFlowPumpData.rssi);
        Serial.print(" snr ");
        Serial.println(gloriaTankFlowPumpData.snr);
    }else  if(packetSize==sizeof(PanchoTankFlowData)){
        LoRa.readBytes((uint8_t*)&panchoTankFlowData,sizeof(PanchoTankFlowData));
        panchoTankFlowData.secondsTime=messageReceivedTime;
        panchoTankFlowData.rssi = LoRa.packetRssi();
        panchoTankFlowData.snr = LoRa.packetSnr();
        Serial.print("Pancho from ");
        Serial.print(panchoTankFlowData.devicename);
        Serial.print(" at:")';'
    }else{
        badPacketCount++;
        Serial.print("  receive bad data size= ");
        Serial.println(packetSize);
    }
}
</pre>

Any suggestions on how to change the code so that R1 receives data from both S1 and S2?  S1 is sending every 10 seconds and S2 is sending every 30 seconds.  Is that too often?

thanks

Kongduino, 2023-07-24:
You should use CAD (Channel Activity Detection) to make sure that the other sender isn't already transmitting: what you are seeing is most probably interferences between both senders, which are at times sending, or trying to, at the same time.

Unfortunately CAD hasn't been implemented in the original library, but I seem to remember someone has a fork with CAD added to it. This is the main weakness of this library – it has been abandoned, mostly, and various people have made an effort to improve it, but these improvements have by and far not been integrated in the original library.

morganrallen, 2023-07-24:
CAD has been merge via #334 but a release has not been done. I called for testing and none of the regulars responded. Additionally in that comment I pointed out, I believe, CAD does not work the way most people are assuming. I wanted to run some tests to confirm this but haven't had the free time (or pay) to get to this.

https://github.com/sandeepmistry/arduino-LoRa/pull/334#issuecomment-1489462169

Kongduino, 2023-07-25:
Oh thanks I wasn't aware (OBVIOUSLY!)... How could I help with testing?

arifainchtein, 2025-02-07:
would love to help as well,  let me know
thanks