sandeepmistry/arduino-LoRa #661: onRecive() method never triggered.

Alain-Godo, 2023-08-14:
Hi everyone!!! I have the following issue. When I receive the data in a sync way all is fine (using the loop for example) but the callback function is never triggered using the onReceive() method.

Board: *ESP32 DOIT DEVKIT V1*
Pin used: *HSPI + 4 (Reset) + 2 (DIO0)*

### Setup code:

hspi = new SPIClass(HSPI);
hspi->begin();
lora2.setSPI(*hspi);

lora2.setPins(15,4,2);
Serial.println("Begining 2...");
Serial.println(lora2.begin(915E6));
lora2.onReceive(lora2Callback);
lora2.receive();

### Callback function:

void lora2Callback(int packet_size) {
   Serial.print("Received packet: '");

    // read packet
    while (lora2.available()) {
    Serial.print((char)lora2.read());
    }

    // print RSSI of packet
    Serial.print("' with RSSI ");
    Serial.println(lora2.packetRssi());
}

qiweimao, 2024-06-03:
Try to keep your callback extremely short. It's a system interrupt. This problem bugged me for days.

Read this issue to solve all your confusions:[https://github.com/sandeepmistry/arduino-LoRa/issues/127](url)

yahyatamim, 2025-01-12:
@qiweimao thanks, you saved my day.