sandeepmistry/arduino-LoRa #667: Any plans to support the 1262 module which is replacing the 1276 for many applications?

factoidforrest, 2023-09-13:
Popular module, in the heltec v3 and the new lilygo both use it, it's better in general.

morganrallen, 2023-09-13:
I would personally like to start exploring it but have no hardware at present. I'm (slowly) designing a new dev board with this goal in mind but contract work has taken priority recently.

dylanjobson, 2023-11-02:
Is there a way to send data from a 1262 using a different library, and receive it on a 1276 using this library?

morganrallen, 2023-11-02:
In theory they should be compatible, it's on my list of things to investigate but I haven't had a chance yet. I'm not aware of any dedicated 1262 libraries.

samograsic, 2024-01-11:
Hi, do you have any updates on this issue? While there are plenty of 1262 libraries available, it would be great to find one that is compatible with this particular library.

Kongduino, 2024-01-11:
> Is there a way to send data from a 1262 using a different library, and receive it on a 1276 using this library?

Yes. I do this all the time. I have a mix of sx1276 and sx1262 devices and they all communicate together without issues.

Kongduino, 2024-01-11:
> In theory they should be compatible, it's on my list of things to investigate but I haven't had a chance yet. I'm not aware of any dedicated 1262 libraries.

There's [Semtech's plain vanilla implementation](https://github.com/Lora-net/sx126x_driver), and [this library](https://github.com/beegee-tokyo/SX126x-Arduino), by a RAKwireless employee, which works on 3 different devices, nRF52840, ESP32 and RPi2040.

The RAK library should be easy to adapt to other models / MCUs.

samograsic, 2024-01-11:
> > Is there a way to send data from a 1262 using a different library, and receive it on a 1276 using this library?
> 
> Yes. I do this all the time. I have a mix of sx1276 and sx1262 devices and they all communicate together without issues.

Have you done any modification to the Sandeepmistry loar library? I tried on a couple of different boards using 1262 and it does not work. Any specific modification needed in how to wire the DIO0-3 signals?

Kongduino, 2024-01-12:
I have but they don't concern transmission. I use mostly ESP32 boards with LoRa sx1276 (sometimes stm32 boards too), and RAK boards (nRF52 or STM32) with sx1262. No issues whatsoever. The setups must be identical though. SF / BW / CR, preamble, etc

zubroid, 2024-05-31:
@Kongduino could you please give some code example how to receive data from 1278(LiLyGo) to 1262(HelTec Lora Tracker)? For transmitting from 1278 I use current library, but for receiving I use RadioLib and I cannot catch packages 😥

kamaradclimber, 2025-01-19:
@Kongduino if you have any example, I think you would make a lot of people happy :)

Kongduino, 2025-01-19:
I missed the previous message. I'll dig around my `Arduino` folder and see what I can post.

kamaradclimber, 2025-05-08:
@Kongduino  were you able to find an example that would work with a 1262 ?