sandeepmistry/arduino-LoRa #683: Starting LoRa with ESP32 fail

phutai9101, 2024-03-05:
hi everybody, I am doing a project about esp32 with module lora sx1278. I can't start the lora module and get an error. Help meeee! thanks for reading.
[image]
[image]

qiweimao, 2024-06-03:
while(true) is causing a inifnite loop, which results in your program crashing. Get rid of that first. Then, check if you can specify MOSI, MISO, SCK.

cagroh, 2024-10-17:
This usually happens by some misconnection to the default SPI MISO, MOSI and SCLK pins and the default values depend on the type of ESP32 you are using.
You can set your own SPI pins by calling spi.begin(LoRa_SPI_CLK, LoRa_SPI_MISO, LoRa_SPI_MOSI, LoRa_SPI_SS); where each pin points to your configuration.
Note this must be called BEFORE you call LoRa.begin().
Cesar