sandeepmistry/arduino-LoRa #689: Starting lora failed

Tad12923, 2024-04-06:
I am new to lora and arduino so I just tried out this code and schematic online but it keep saying starting lora failed 
Here's the code

3
4
5
6
7
8
9
10
11
12
13
14
15
16
17
18
19
20
21
22
23
24
25
26
27
28
29
30
31
32
33
34
35
36
37
38
39
40
41
42
43
44
45
46
47
48
49
50
51
52
53
54
55
56
57
58
59
60
61
62
63
64
65
66
67
68
69
70
71
72
73
74
75
76
77
78
79
80
81
82
83
84
85
86
87
88
89
90
91
92
93
94
95
96
97
98
99
100
101
102
#include <SPI.h>
#include <LoRa.h>
#include <Wire.h>
#include <Adafruit_GFX.h>
#include <Adafruit_SSD1306.h>
#include <DHT.h>
 
#define SCREEN_WIDTH 128 // OLED display width, in pixels
#define SCREEN_HEIGHT 64 // OLED display height, in pixels
 
// Declaration for an SSD1306 display connected to I2C (SDA, SCL pins)
#define OLED_RESET     LED_BUILTIN // Reset pin # (or -1 if sharing Arduino reset pin)
Adafruit_SSD1306 display(SCREEN_WIDTH, SCREEN_HEIGHT, &Wire, OLED_RESET);
 
#define DHTPIN 0          //pin where the dht11 is connected
#define ss 15
#define rst 16
#define dio0 2
int counter = 0;
 
DHT dht(DHTPIN, DHT11);
void setup() 
{
  Serial.begin(115200);
  dht.begin();
 
  // SSD1306_SWITCHCAPVCC = generate display voltage from 3.3V internally
  if(!display.begin(SSD1306_SWITCHCAPVCC, 0x3C)) { // Address 0x3D for 128x64
    Serial.println(F("SSD1306 allocation failed"));
    for(;;); // Don't proceed, loop forever
  }
  
  while (!Serial);
  Serial.println("LoRa Sender");
  LoRa.setPins(ss, rst, dio0);
    if (!LoRa.begin(433E6)) {
    Serial.println("Starting LoRa failed!");
    delay(100);
    while (1);
  }
  
  display.display();
  delay(2);
  display.clearDisplay();
}
 
void loop() 
{
  float h = dht.readHumidity();
  float t = dht.readTemperature();
 
  if (isnan(h) || isnan(t)) 
  {
  Serial.println("Failed to read from DHT sensor!");
  return;
  }
  Serial.print("Temperature: ");
  Serial.print(t);
  Serial.print(" degrees Celcius, Humidity: ");
  Serial.println(h);
  Serial.println();
  Serial.print("Sending packet: ");
  Serial.println(counter);
 
  display.clearDisplay();
  display.setTextSize(2);             // Normal 1:1 pixel scale
  display.setTextColor(WHITE);        // Draw white text
  display.setCursor(0,0);             // Start at top-left corner
  display.print(F("Pkt No:"));
  display.print(counter);
  
  display.setCursor(0,20); 
  display.print("Tem:");
  display.print(t);
  display.print("C");
 
  display.setCursor(0,40); 
  display.print("Hum:");
  display.print(h);
  display.print("%");
  display.display();
  
  // send packet
  LoRa.beginPacket();
  LoRa.print(F("Pkt No:"));
  LoRa.println(counter);
  
  LoRa.print("Temp: ");
  LoRa.print(t);
  LoRa.println("°C");
  
  LoRa.print("Hum:  ");
  LoRa.print(h);
  LoRa.print("%");
  LoRa.println("");
  
  LoRa.endPacket();
 
  counter++;
 
  delay(3000);
}
[image]
I am using esp8266 node MCU and lora-ra2

CodingMommy, 2024-05-03:
Sounds silly but have you connected a 433mhz suitable antenna to the LoRa module?

sercero, 2024-05-19:
The antenna issue is important, without the antenna you won't be able to transmit.
But "lora failed" is not an antenna problem, rather something wrong with the module most likely the connection.

Are you powering the Ra-02 module from the NodeMCU?

You should use an external power supply because the module can draw 200mA of power.

Check the docs: https://docs.ai-thinker.com/en/lora/man

Another thing: if things are not working you should simplify your setup and just connect the NodeMCU to the Ra-02 module remove the display and DHT sensor for now.