sandeepmistry/arduino-LoRa #691: LORA COMMUNICATION USING NANO

Srivardhan06, 2024-05-18:
I'm working on lora communication using nano i m not receving the data 
and some times i'm receving the garabage values like:????????????
can you help me in this issue 

thank you

sercero, 2024-05-19:
Huh?

Please tell more about your setup, what are you using. How did you connect everything?

Which code are you using? The sender and receiver samples?

Srivardhan06, 2024-05-19:
IM USING NANO AND LORA SX1278 RA1 

**SENDER CODE :**

#include <SPI.h>
#include <LoRa.h>

byte msgCount = 0;               // Count of outgoing messages
byte localAddress = 0xBB;        // Address of this device
byte destination = 0xFF;         // Destination to send to
long lastSendTime = 0;           // Last send time
int interval = 3000;              // Interval between sends

/*  
 *   Transmitter side Code
 * 
  Module SX1278 // Arduino UNO/NANO    
    Vcc         ->   3.3V
    MISO        ->   D12
    MOSI        ->   D11     
    SLCK        ->   D13
    Nss         ->   D10 
    GND         ->   GND    
 */

void setup() {
  Serial.begin(115200);
   
  while (!Serial);  
  Serial.println("LoRa Sender");
  
  if (!LoRa.begin(433E6)) { // or 915E6, the MHz speed of your module
    Serial.println("Starting LoRa failed!");
    while (1);
  }
}

void loop() {
String MyMessage = "Hello World";
  LoRa.beginPacket();  
  LoRa.print(MyMessage);
  LoRa.endPacket();
  delay(300);
 }
 
 RECEIVER CODE :
 
#include <SPI.h>
#include <LoRa.h>

byte data[50];
byte msgCount = 0;               // Count of outgoing messages
byte localAddress = 0xFF;        // Address of this device
byte destination = 0xBB;         // Destination to send to
long lastSendTime = 0;           // Last send time
int interval = 300;              // Interval between sends

/* 
 ***  Receiver Side Code**

#include <SPI.h>
#include <LoRa.h>  
String inString = "";    // string to hold incoming charaters
String MyMessage = ""; // Holds the complete message

void setup() {
  Serial.begin(115200);
 
  while (!Serial);
  Serial.println("LoRa Receiver");
  if (!LoRa.begin(433E6)) { // or 915E6
    Serial.println("Starting LoRa failed!");
    while (1);
  }
}

void loop() {
  
  // try to parse packet
  int packetSize = LoRa.parsePacket();
  if (packetSize) { 
    // read packet    
    while (LoRa.available())
    {
      int inChar = LoRa.read();
      inString += (char)inChar;
      MyMessage = inString;       
    }
    inString = "";     
    LoRa.packetRssi();    
  }
      
  Serial.println(MyMessage);  
  
}

Srivardhan06, 2024-05-19:
I'M Connecting the pins from sx`1278 to nano 

and i'm providing the external power supply for lora 

Module SX1278 // Arduino UNO/NANO
Vcc -> 3.3V
MISO -> D12
MOSI -> D11
SLCK -> D13
Nss -> D10
GND -> GND

Srivardhan06, 2024-05-19:
I'M using nano for both transmitter and receiver side

sercero, 2024-05-19:
Tomorrow I'll get those modules 😊

We might have very similar setups...

Can you show a picture of how you connected everything?

Another thing... why don't you print the RSSI of the received packet?
That might help...

Srivardhan06, 2024-05-19:
OK TQ

Srivardhan06, 2024-05-19:
Connection are same as i said before |

Module SX1278 // Arduino NANO
Vcc -> 3.3V
MISO -> D12
MOSI -> D11
SLCK -> D13
Nss -> D10
GND -> GND
DI00 -> D2
RST -> RST

vcc pin is connected to 3.3v nano board 
MISO pin is connected to D12 pin

sercero, 2024-05-19:
I wanted to see how you did the physical connection in part to help and in part to see how to do it myself because that module I think that the holes do not align with a protoboard.

Srivardhan06, 2024-05-20:
ok i will share

Srivardhan06, 2024-05-20:
[image]

Srivardhan06, 2024-05-20:
I have doubt should i use level converters between lora and nano

Srivardhan06, 2024-05-20:
[image]

Srivardhan06, 2024-05-20:
[image]

Srivardhan06, 2024-05-21:
I HAVE GIVEN COMMON GROUND FOR EXTERNAL 3.3 V SUPPLY

Srivardhan06, 2024-05-21:
the ground  from nano is given to the GROUND LINE of the bread board and ground from the external 3.3v is connected to the bread board ground line and  lora is also connected to same ground line

sercero, 2024-05-21:
Have you checked continuity between the soldered cables?

I soldered cables to that module yesterday and it is a pain in the a**.

Perhaps there is a short circuit somewhere.

Srivardhan06, 2024-05-24:
Haa the connection are right 
 
I have checked the continuity

Srivardhan06, 2024-05-24:
Can i know that you are able to receive the data from lora using nano

Srivardhan06, 2024-05-24:
If  it is working can you share the circuit diagram

sercero, 2024-05-24:
[image]

Hello, I was able to make it work.

But I'm using Arduino Pro Mini.

These are the connections:
```
ARDUINO --------- LoRa Module
GND <- NEGRO ---> GND
VCC <- ROJO ----> 3.3V
10  <- MARRON --> NSS
11  <- AZUL ----> MOSI
12  <- NARANJA -> MISO
13  <- BLANCO --> SCK
9   <- VIOLETA -> RST
2   <- VERDE ---> DIO0
```

The coil antenna seems to be pretty bad, -80dBm with two modules next to each other.

I get better reception with the modules Ra-02 and a cable as an antenna

Srivardhan06, 2024-05-25:
can you check with Aduino nano

Srivardhan06, 2024-05-25:
[image]

Srivardhan06, 2024-05-25:
[image]

Srivardhan06, 2024-05-25:
and can you share the outputs  and also print rssi value of received packet

sercero, 2024-05-25:
Hello, I don't have an Arduino nano so that test is not possible for me...

Is it possible that Arduino nano uses 5V?

You should use 3,3V for the LoRa modules, they might work at 5V but long term they might be damaged by that voltage.

The RSSI I'm getting is -64dBm with the modules next to each other.

sercero, 2024-05-25:
> Connection are same as i said before |
> 
> Module SX1278 // Arduino NANO Vcc -> 3.3V MISO -> D12 MOSI -> D11 SLCK -> D13 Nss -> D10 GND -> GND DI00 -> D2 RST -> RST
> 
> vcc pin is connected to 3.3v nano board MISO pin is connected to D12 pin

I see you connected RST -> RST, shouldn't it go to D9?

Srivardhan06, 2024-05-25:
haa yes rst---> d9

Srivardhan06, 2024-05-25:
i m using logic level converters between lora and nano 
but i m getting

 
LoRa Receiver
Starting LoRa failed!

Srivardhan06, 2024-05-26:
and  can you share the code and as well as ouput

sercero, 2024-05-26:
I took the code from here: https://iotprojectsideas.com/multiple-lora-nodes-communication-with-master-lora-node/

The output won't tell you much.

Srivardhan06, 2024-05-27:
can you give me the voltage at every pins of lora at sender and receiver

Srivardhan06, 2024-05-27:
Hoo oky i will once again

Srivardhan06, 2024-05-27:
And also i m receiving the data  but there is a small issue that when i m sending 10 packets of data i'm only receiving 7 to 8 packets with -54 Rssi  

I think there is a power supply  issue with lora

sercero, 2024-05-27:
How often are you sending the packets?

The LoRa protocol is rather limited in badwitdh

Srivardhan06, 2024-05-31:
Is anything we need to configure  for lora

ErfanM98, 2024-08-18:
I have a similar problem. Did you find any solutions? thanks