sandeepmistry/arduino-LoRa #696: Arduino Uno + RFM95W sending/receiving issue

remi-duclos, 2024-06-30:
Hello,

I'm having trouble getting RFM95W to work with Arduino Uno and my problems are the following : 
**- On the sender setup, sending packets stop at 0, 1 or 2** (LoraSender example, Image1 below). I tried to measure the input voltage of the RFM95W during transmission and it was stable at 3.27V. **Strangely, when I measured the voltage, the sending packet didn't stop at 0, 1 or and kept going until I removed the probes of my multimeter from the RFM95W** (Image2 below).
**- On the receiver setup, there were 3 cases: 1) Nothing happens, serial monitor stays clear. 2) I'm receiving the packet with very low RSSI (-157)** (Image3 below). **3) I'm receiving the packet with low RSSI (-109) and the message keeps spamming in the serial monitor** (Image4 below).

**Image1:**
[image]

**Image2:**
[image]

**Image3:**
[image]

**Image4:**
[image]

 I have the following setup for both sender and receiver :
- Arduino Uno board with screw shield 
- Buck converter (5V to 3.3V to power the RFM95W) connected to Arduino 5v output and RFM95 3.3v input
- 8 canals 5V/3.3V bidirectionnal logic level converter
- RFM95W

Here are a scheme and a picture of the setup (antenna in the scheme is not representative of my real antenna, which is a helical antenna as you can see on the picture) :
[image]
[image]

Each RF95W are connected to Arduino Uno following @sandeepmistry instructions :
| RFM95W  | Arduino Uno |
| ------------- | ------------- |
| VCC  | 3.3V (from buck converter)  |
| GND  | GND (from buck converter)  |
| SCK | 13  |
| MISO | 12 |
| MOSI | 11 |
| NSS | 10 |
| NRESET | 9 |
| DIO0  | 2 |

And I'm using the examples of Sender and Receiver supplied by @sandeepmistry, where I've only changed the frequency (868 MHz instead of 915 MHz).

**Sender :**
```
#include <SPI.h>
#include <LoRa.h>

int counter = 0;

void setup() {
  Serial.begin(9600);
  while (!Serial);

  Serial.println("LoRa Sender");

  if (!LoRa.begin(868E6)) {
    Serial.println("Starting LoRa failed!");
    while (1);
  }
}

void loop() {
  Serial.print("Sending packet: ");
  Serial.println(counter);

  // send packet
  LoRa.beginPacket();
  LoRa.print("hello ");
  LoRa.print(counter);
  LoRa.endPacket();

  counter++;

  delay(5000);
}
```

**Receiver :**
```
#include <SPI.h>
#include <LoRa.h>

void setup() {
  Serial.begin(9600);
  while (!Serial);

  Serial.println("LoRa Receiver");

  if (!LoRa.begin(868E6)) {
    Serial.println("Starting LoRa failed!");
    while (1);
  }
}

void loop() {
  // try to parse packet
  int packetSize = LoRa.parsePacket();
  if (packetSize) {
    // received a packet
    Serial.print("Received packet '");

    // read packet
    while (LoRa.available()) {
      Serial.print((char)LoRa.read());
    }

    // print RSSI of packet
    Serial.print("' with RSSI ");
    Serial.println(LoRa.packetRssi());
  }
}
```
I don't understand why the sending setup is working when I'm putting the multimeter probes to 3.3V and GDN pins of the RFM95W. And why is the RSSI so low while both setups are 1 meter apart.

cagroh, 2024-10-17:
Hi Remi!
Have you succeed in solving the issue?
I had some issues regarding low RSSI when I had the antenna not properly soldered to the RFM95 ANA pin.
Cesar

remi-duclos, 2024-10-17:
> Hi Remi! Have you succeed in solving the issue? I had some issues regarding low RSSI when I had the antenna not properly soldered to the RFM95 ANA pin. Cesar

Hi Cesar, I couldn't solve the problem even after resoldering the antenna and the other wires several times... I ended up buying Maduino LoRa boards and I'm communicating in serial between the Arduino and the Maduino LoRa board.

cagroh, 2024-10-18:
That's too bad...
I'm giving up on sandeepmistry's library altogether and giving RadioLib a try.
Let's see how it goes. If nothing works, I might try it the hard way...