sandeepmistry/arduino-LoRa #703: Usage of int in LoRa library

goldyrathore, 2024-09-25:
The LoRa library is using int at several places. Int is 2 bytes on Arduino whereas the intended size probably was 1 byte and not 2. Isnt it a bug to use int in place of 8 bits. For example "int packetRssi();" returns 2 bytes though it needs to return only 1 byte.

Kongduino, 2024-09-25:
Not quite. The RSSI value is a signed number, and can be below -128. How are you going to store that in a byte...?

goldyrathore, 2024-09-26:
What about other places like  return value of LoRaClass::beginPacket?  And several other places where a uint8_t would just do fine

Kongduino, 2024-09-26:
I remember a discussion a long time ago here where Sanddep said he didn't want to use types like `uint8_t`. Of course `byte` would have been fine but anyway, it keeps things uniform.

morganrallen, 2024-09-26:
FWIW, I believe (but could be wrong, maybe it documented somewhere? probably not), I believe using `int` universally is the 'Arduino way'.

closed 2024-09-26 by morganrallen