sandeepmistry/arduino-LoRa #716: Bug in packetFrequencyError

Nick507, 2025-03-09:
In function LoRaClass::packetFrequencyError()
there is a wrong conversion: freqError -= 524288;
it must be freqError -= 1048576;
because 1<<20 = 1048576

torntrousers, 2025-03-09:
Hello @Nick507, I added that packetFrequencyError function, [back in 2018](https://github.com/sandeepmistry/arduino-LoRa/commit/27e2963c9710fd01523a1dce11f511b1cd368eca), so i don't remember the code super well! I think it might have been derived from code from a Semtech library. Could you give any more explanation on how you found the problem or why you think its a bug?

Nick507, 2025-03-10:
Hello Anthony, according to sx127x datasheet, frequency error value is 20 bit two's complement value. The algorithm of converting negative value from two's complement form is to substract 2^N from that value, where N is number of bits, which is 20 in this case.
https://en.wikipedia.org/wiki/Two%27s_complement
524288 is the 2^19.
The simple example:
1111 - is -1 for 4bit two's complement value (according to wiki)
most significant bit is set, it means that this is negative value, and need to convert it: 15-16=-1, where 
15=1111 (our value)
16=2^4=1<<4

Nick507, 2025-06-03:
Sorry for messing, there is an additional masking for MSB byte in code:
readRegister(REG_FREQ_ERROR_MSB) & 0b111
So result will be the same, just strange transformation algorithm.

closed 2025-06-03 by Nick507