sandeepmistry/arduino-LoRa #718: Method that should be added.

Gabriel192K, 2025-03-24:
Hello, I was using this great library for quite some time but I never got around the fact it lacks a very basic method i.e. function. The method I imply should return some sort of device ID or chip version. This would be a great and simple way to check if the device is alive and/or connected. I saw you already used this but as a local check inside the be ```begin()``` function. So my whole point is to add a public method called ```version()``` or something similar to read that device ID. I'm already using this modified library and it works like a charm.

waltje, 2025-04-26:
I would name that getDeviceId()  - version() would more relate to the library's version imho.

Gabriel192K, 2025-04-27:
@waltje yes, you’re right. The name is irrelevant as the point is to have a public method to access that register.